OpenWonderLabs/SwitchBotAPI-BLE #5: Lock BLE API

AdyRock, 2022-06-11:
### Current Situation

There is no BLE API documentation for the lock API except for the device id

### Proposed Change

Add documentation for the Lock BLE API

### Additional Context

It is desirable to integrate the lock into other home automation systems.

kaxio, 2022-07-12:
Hey team, any updates on this?

Jacksonbm1, 2022-07-18:
I saw a post in a live chat with a cofounder of switchbot that said the ble lock api is coming:

Here's the op:

"Hey. It's Richard here. I am one of the cofounders of
this little robot SwitchBot. I am more on sales and
marketing, but i also know how to code and play
around with JavaScript and Python myself. I have
around 160+ connected device, 20-30 scenes/
routines/automations or whatever you call it at my
place and most of them are in use, which means I
kinda know what smart home is. Also I use Alexa
most of the time at home and Siri when I am on my
phone. Just wanna say hi and if you guys have
anything you wanna tell me, just let me know. Would
like to interact with you guys more from now on. In
case i missed you comments, pin me again. I am
also on Twitter in case you wanna talk there too.
https://twitter.com/richardmou63/"

Here's the reply that says the lock api is coming:

"1. Lock API is definitely on its way. We gotta
test the security level before releasing and
there is no ETA yet."

Post: https://www.reddit.com/r/TrySwitchBot/comments/vzr3y6/first_post_hello_world/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

brandon-dacrib, 2022-08-31:
I don't understand. Why was this issue closed even though there is still no documentation on using the switchbot lock via BLE?

Jacksonbm1, 2022-08-31:
It's listed as open for me? But I believe comments may re open it. Pretty sure GitHub auto closes issues that don't get updates and go stale

martindinicolino, 2022-09-25:
with V1.1 of the API dropping about a week ago and bringing back the lock and unlock capability to the webAPI (which is awesome, thank you very much for that), is there any timetable in regards to this functionality being added to the bleAPI?

Jacksonbm1, 2022-09-26:
Best way I can find is to message Richard Mou on twitter. He responds quickly

r-glebov, 2022-10-14:
Looking forward for the BLE API support!

martindinicolino, 2022-10-14:
> Best way I can find is to message Richard Mou on twitter. He responds quickly

i‘m not on twitter, could somebody do that please?

Minsheng, 2022-11-15:
> ### Current Situation
> There is no BLE API documentation for the lock API except for the device id
> 
> ### Proposed Change
> Add documentation for the Lock BLE API
> 
> ### Additional Context
> It is desirable to integrate the lock into other home automation systems.

This is Davidson from the dev team. We are working on the BLE API and we will update the BLE API doc hopefully in the next month. I myself will oversee the progress and keep you posted. Thanks a lot for being patient and supportive! You guys are the best!

Minsheng, 2022-11-21:
> > This is Davidson from the dev team. We are working on the BLE API and we will update the BLE API doc hopefully in the next month. I myself will oversee the progress and keep you posted. Thanks a lot for being patient and supportive! You guys are the best!
> 
> This is fantastic news! Out of curiosity, what features will the api include?
> 
> * ability to lock/unlock
> * ability to read state of lock
> * ability to read state of door
> * ability to read state of battery
> 
> Best

We plan to support these triggers first,
- ability to read state of lock
- ability to read state of door
- ability to read state of battery

The actions will have to come after.

Davidson

marcomow, 2022-11-21:
awesome! @Minsheng do you have an estimated timeline for both reading and actions? Asking as our company is deciding which product to choose, so it may be worth to wait, but we need to understand how long. Thank you!

sonny-chen, 2022-11-22:
> This is fantastic news! Out of curiosity, what features will the api include?
> 
> * ability to lock/unlock
> * ability to read state of lock
> * ability to read state of door
> * ability to read state of battery

Please also add the ability to set PIN. This will create a huge use case for vacation homes.

Minsheng, 2022-11-25:
> awesome! @Minsheng do you have an estimated timeline for both reading and actions? Asking as our company is deciding which product to choose, so it may be worth to wait, but we need to understand how long. Thank you!

We have released the broadcast definition for Lock. The pull request is being reviewed atm.
We don't have a timeline for the implementation of Lock actions yet. Please stay tuned.

federicotravaini, 2022-12-12:
Just goona buy in advance the lock waiting for this integration to be live asap 😃

advorsky73, 2023-01-01:
Happy New Year ;) Any updates?

donavanbecker, 2023-01-01:
@advorsky73 check this repo out: https://github.com/Danielhiversen/pySwitchbot

advorsky73, 2023-01-02:
@donavanbecker thats for cloud api, how does this help for BLE ??? 🤔

donavanbecker, 2023-01-02:
@advorsky73 pretty sure this is all BLE: https://github.com/dsypniewski/pySwitchbot/blob/ed46ac93e2c8985037e21f31e4bb7b933e17714f/switchbot/devices/lock.py

dsypniewski, 2023-01-02:
@advorsky73 it's all BLE, not cloud. I did some investigating of messages between the lock and the app. I'll try to make a PR here once I'll finish up and organize my notes.
Next release of HA should also have this integrated, it already is in the beta.

jpconfessor, 2023-01-05:
FINALLY!! Home assistant added support for this lock via BLE! check version 2023.1 ;) Works fine and exposes several states:
[image]

martindinicolino, 2023-01-05:
@jpconfessor beautiful! thanks for letting us know 👍

are we sure locking and unlocking work over BLE though? I thought that wasn‘t yet available vie the BLE API, but i‘d love to be corrected

this is a big improvement as is, but it‘s worth even more if it‘s locally controlled

dsypniewski, 2023-01-05:
@martindinicolino Yes, as I said before it's local. I spend about a month reverse engineering the lock communication and making the integration for Home Assistant, that's why the integration works while there aren't any action related docs released by SwitchBot yet.
The only internet connection is needed during setup to retrieve the locks encryption key from SwitchBot internal API.

martindinicolino, 2023-01-05:
@dsypniewski this is huge! thank you for your effort

i‘ll give it a try as soon as i get home

jpconfessor, 2023-01-05:
> 

yes! I was able to validate using my ble proxy logs (from esphome). 

I’ve noticed that the very first command after a while requires a connection, and due to this it takes around 1s to lock/unlock… if you try again.. it is almost instant….
so, this very first command is still waaaay faster than api (I was using the smartthings integration.. so, even worst!)

I’ve watched the release video from HA and they’ve mentioned some improvements around this area in the new version of esphome that requires a serial update (not OTA) in order to give us full performance. I will give it a try to see if I can reduce this even more :)

setup:
I am using home assistant 2023.1 with the switchbot integration via bluetooth proxies around my house (esp32 devices - m5 atom lite) - check this: https://esphome.github.io/bluetooth-proxies/

kklicker777, 2023-01-05:
I have also set this up in Home Assistant and can confirm it is local.  You should be proud of your work here.  I know I read a lot of comments about hesitation in buying this lock due to the lack of local support on Home Assistant.  I would expect sales of this lock to improve greatly due to this improvement.  Great job.

andreasbuff, 2023-01-05:
> yes! I was able to validate using my ble proxy logs (from esphome).
>
> I’ve noticed that the very first command after a while requires a connection, and due to this it takes around 1s to lock/unlock… [...]

Works great, as described! Thanks everyone for this solution! Regards

Write, 2023-01-07:
> @martindinicolino Yes, as I said before it's local. I spend about a month reverse engineering the lock communication and making the integration for Home Assistant, that's why the integration works [...]

Excellent work. Works flawlessly, thanks you so much ! 
Do you have any donations link ? That's the kind of work that definitely deserve it.

donavanbecker, 2023-01-07:
@jpconfessor, I tried out the home assistant component, but my SwitchBot account doesn't have a password since I use Login in with Apple. Is there anyway to authenticate with the token and secret?

Write, 2023-01-07:
> @jpconfessor, I tried out the home assistant component, but my SwitchBot account doesn't have a password since I use Login in with Apple. Is there anyway to authenticate with the token and secret?

You’d have to unpair everything, delete your account and recreate one with an email. That’s what I did.
The other way to get the token still require both an email and password so I don’t think there’s a way around it.

jpconfessor, 2023-01-07:
> 

In order to get the encryption key you still need the username and password since the python script will request it
https://github.com/Danielhiversen/pySwitchbot#obtaining-locks-encryption-key

dsypniewski, 2023-01-07:
> Excellent work. Works flawlessly, thanks you so much ! Do you have any donations link ? That's the kind of work that definitely deserve it.

@Write Thank you, I've put some links in my profile as well as here:
https://paypal.me/DamianSypniewski
https://revolut.me/damiansypniewski

> @jpconfessor, I tried out the home assistant component, but my SwitchBot account doesn't have a password since I use Login in with Apple. Is there anyway to authenticate with the token and secret?

@donavanbecker unfortunately currently only normal (non SSO) accounts can be used no matter if it's through HA or the script, but I'll be looking into trying to get those accounts working as well.

donavanbecker, 2023-01-07:
Thanks @dsypniewski!

dsypniewski, 2023-01-08:
@donavanbecker could you try the web auth option in the script from my branch ?https://github.com/dsypniewski/pySwitchbot/tree/lock-key-script-web-auth

slight316, 2023-01-08:
Hello,
I tried to integrate with home assistant via a bluetooth dongle.  Dongle is directly connected to a Dell Wyse 3030 (no wifi)
hardware info:
[image]

The lock is Discovered but when I go to configure with my encryption key or my username/password I get Unknown error occurred.

[image]

Any suggestions?

donavanbecker, 2023-01-08:
> @donavanbecker could you try the web auth option in the script from my branch ?https://github.com/dsypniewski/pySwitchbot/tree/lock-key-script-web-auth

@dsypniewski, I must do this on linux? can't do it on Mac?

dsypniewski, 2023-01-09:
@donavanbecker the script needs to register a custom URL handler to receive the authentication results from the browser and each OS has different method of doing this so currently only Linux is supported. I don't have a Mac to test, but I'll try looking into how to do it there.

ashald, 2023-01-09:
Ah, and since it's the Switchbot official repo - just for the record.
1. I ordered the lock ASAP once it was announced since I thought it'd have local BLE interface available for 3rd parties right away
2. Once I learned it doesn't I canceled the order
3. Once community reverse-engineered the interface and built integration for HomeAssistant, I bought a lock right away, super happy with the product now, and consider buying 1 or 2 more. I won't claim everyone is like me, but I'd bet that I'm by far not alone. That is to highlight how important the full BLE protocol available to your community.
Thanks everyone!

slight316, 2023-01-09:
on a somewhat related but not related tangent.  Any chance that the keypad operation (adding keys, deleting keys, etc) can also be integrated into home assistant?

dsypniewski, 2023-01-09:
@slight316 With enough time in theory it should be possible to even support the whole setup of lock/keypad but it takes a lot more time to investigate the multi step processes compared to single step like lock/unlock.
As much as I'd like to have a complete replacement for the SwitchBot app I don't think it's going to happen any time soon.

slight316, 2023-01-09:
@dsypniewski I appreciate your candour and understand Rome wasn't built in a day and completely appreciate all the work that has been done to reverse engineer the lock.  

It would be brilliant if the keypad was reverse engineered as well.  If only to allow me to manage keys remotely through HA instead of having to be within BLE distance to the keypad.

dsypniewski, 2023-01-10:
@slight316 I agree, and will definitely try just don't know when or how feasible it will be.

@donavanbecker I just found out that there is a docker image for macOS vm, the chances of getting this working on Mac just went up considerably.

donavanbecker, 2023-01-10:
@D

slight316, 2023-01-13:
@dsypniewski I find myself with a little bit of spare time.  Is there anyway that I can help with reverse engineering the keypad functionality?  I don't have much experience in this department but if you can send me towards some articles or something I'm eager to learn.

dsypniewski, 2023-01-15:
@slight316 As to sniffing bluetooth traffic I can recommend series of videos mentioned here https://hackaday.com/2021/03/23/a-crash-course-on-sniffing-bluetooth-low-energy/
This requires a dongle that allows sniffing Bluetooth - you can usually buy one pretty cheap on sites like AliExpress but I didn't want to wait so I made mine from nRF52840 dongle flashed with this https://www.nordicsemi.com/Products/Development-tools/nRF-Sniffer-for-Bluetooth-LE
What also helped a lot was looking through decompiled java code and string resources of the Android SwitchBot app, this article https://medium.com/helpshift-engineering/reverse-engineer-your-favorite-android-app-863a797042a6 seems to be using similar set of tools that I was using except the last step where I use `fernflower`.
 
@donavanbecker I pushed an update to the branch mentioned earlier that adds support for macOS and Windows could you check it out?
This might require updating `python3` to newer version because I had some problems with the default version provided by macOS.

KiraPC, 2023-01-20:
> @slight316 As to sniffing bluetooth traffic I can recommend series of videos mentioned here https://hackaday.com/2021/03/23/a-crash-course-on-sniffing-bluetooth-low-energy/ This requires a dongle [...]

Hello @dsypniewski I'm trying to test the script on my mac, do I have to install the URL handler? because chrome can't found it.

dsypniewski, 2023-01-20:
@KiraPC while running the script it should create an app that handles those authentication URLs and removes it after the script exits. While the script is running you can check in the `/Applications` directory there should be `switchbot URL handler.app`.

KiraPC, 2023-01-20:
Yes,

I did a little bit of reverse engineering and found.

Anyway it doesn’t want works.

I also tried to pack an app with py2app that listen on url as yours.

It works if I call the app directly from the browser typing switchbot:// and works (I also started a listener for debugging) but not when is cognito to call the app.

Could you help me?

donavanbecker, 2023-01-20:
> @donavanbecker I pushed an update to the branch mentioned earlier that adds support for macOS and Windows could you check it out? This might require updating `python3` to newer version because I had some problems with the default version provided by macOS.

Sorry @dsypniewski, been busy. sorry for not knowing much about python. Want to learn more about it, but how do I run the script? I have the zip downloaded and I tried to execute it using `python3 switchbot_get_lock_key.py`

but then I receive this:
```
Traceback (most recent call last):
  File "/Users/xxxxxx/xxxxxxxxxxx/pySwitchbot-lock-key-script-web-auth/scripts/switchbot_get_lock_key.py", line 6, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

dsypniewski, 2023-01-28:
@KiraPC Do you have any ideas on how this could be improved? This is my first time working with an app for Mac.

@donavanbecker Sorry for the delay but I've also been busy recently - the script is not standalone and needs the pySwitchbot and other libraries installed to work so you need to install the package using pip:
```
pip install git+https://github.com/dsypniewski/pySwitchbot.git@lock-key-script-web-auth
```
If `pip` is not found or not working you might need to use `pip3` instead.
This will also install the script as executable, so there should be no need to use the downloaded one.

donavanbecker, 2023-01-28:
No problems! and thanks @dsypniewski. This is what I received:

```
Registering switchbot:// url handler
Opening auth page
Waiting for the auth response.
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.11/bin/switchbot_get_lock_key.py", line 95, in <module>
    web_auth()
  File "/Library/Frameworks/Python.framework/Versions/3.11/bin/switchbot_get_lock_key.py", line 49, in web_auth
    listener = Listener(address)
               ^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/connection.py", line 447, in __init__
    self._listener = SocketListener(address, family, backlog)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/connection.py", line 590, in __init__
    self._socket.bind(address)
OSError: [Errno 48] Address already in use
``

dsypniewski, 2023-01-28:
@donavanbecker Hmm, this error message is not very helpful but there was one thing that I could think of and I pushed an update, could you check if that fixes your problem?

KiraPC, 2023-01-29:
> @donavanbecker Hmm, this error message is not very helpful but there was one thing that I could think of and I pushed an update, could you check if that fixes your problem?

This was the same error i had. 

I pushed [here](https://github.com/KiraPC/switchbot-web-auth-macos) the standalone application to be installed manually.

Try it and let me now.

> @KiraPC Do you have any ideas on how this could be improved? This is my first time working with an app for Mac.

@dsypniewski Check my repository. The code is basically equals to yours, but it is standalone.

donavanbecker, 2023-01-30:
@KiraPC when I run the commands, I get this:

```
The 'rumps' distribution was not found and is required by the application
```

KiraPC, 2023-01-30:
> @KiraPC when I run the commands, I get this:
> 
> 
> 
> ```
> 
> The 'rumps' distribution was not found and is required by the application
> 
> ```

Damn! 

I stared from an old project, and I forget to remove a dependecy.

In setup.py file remove the "install_requires" line and check again if it compile.

I'm from my smartphone, tomorrow I'll fix on the repo.

donavanbecker, 2023-01-30:
still getting:

```
ImportError: No module named 'rumps'
```
No rush, I can wait till tomorrow.

KiraPC, 2023-01-30:
> still getting:
> 
> 
> 
> ```
> 
> ImportError: No module named 'rumps'
> 
> ```
> 
> No rush, I can wait till tomorrow.

Mmm, I'll test it on a virtual environment tomorrow.

dsypniewski, 2023-01-31:
@donavanbecker how about changes in my branch I mentioned earlier, does that work for you?

KiraPC, 2023-01-31:
@donavanbecker fixed it. Should work now!

federicotravaini, 2023-01-31:
Sorry to interrupt your conversation but maybe you can help with this related issue: https://github.com/home-assistant/home-assistant.io/issues/25859

KiraPC, 2023-01-31:
> Sorry to interrupt your conversation but maybe you can help with this related issue: [home-assistant/home-assistant.io#25859](https://github.com/home-assistant/home-assistant.io/issues/25859)

Hello @federicotravaini I'm using the switchbot Lock too. It's working fine.

I read the 3rd you posted and seems strange.

Could you report your system? Are you using a BLE adapter?

[11 more replies]

closed 2026-10-30 by donavanbecker