OpenWonderLabs/SwitchBotAPI-BLE #11: Curtain BLE designation of dual vs single curtain setup

apollo1220, 2022-10-06:
### Current Situation

Currently there doesn't seem to be a way to know whether the curtain device is setup just as a single curtain or is paired with another curtain device to make a dual curtain.

### Proposed Change

Add information to the GET API which designates which way the curtain device is configured.

### Additional Context

This is for Home Assistant's integration with SwitchBot.

bdraco, 2022-10-06:
We likely need a way in the advertisement data to know which commands to send

If its a dual setup we likely need to send the original commands that https://github.com/Danielhiversen/pySwitchbot/pull/122/files changed.

If its a single setup we need to use the commands that are in there right now

donavanbecker, 2022-10-06:
If you use with OpenAPI then you can see if it is paired with another curtain. That's what I do in the homebridge plugin.

bdraco, 2022-10-06:
Unfortunately Home Assistant only talks to these over Bluetooth so it has no credentials to access the cloud apis

bdraco, 2022-10-16:
Our solution was to send both commands which I assume is successful since the issues have not been reopened https://github.com/Danielhiversen/pySwitchbot/pull/127