OpenWonderLabs/SwitchBotAPI-BLE #15: Advertise state changes instead of needing to ask for SCAN_RSP

bdraco, 2022-12-10:
### Current Situation

When a contact sensor or other changes state, it would be great if it advertise the change immediately instead of having to ask for a scan response.  This would make the Home Assistant work with passive scanning and reflect changes right away which would allow us to solve https://github.com/home-assistant/core/issues/82795

Currently the scan response data can be stale since there is a race between when the contact sensor changes state and the scan response is received which leads to stale and incorrect states being seen.

### Proposed Change

See above

dcousens, 2022-12-16:
@bdraco, I am using passive scanning only for my SwitchBot contact sensors.

The manufacturer data broadcast seems to include motion and contact flags, and two 16-bit timers, respectively, counting since their last state change (~`256/second`).  There is a 4-bit counter too at the end of the message for built-in button presses.

The motion sensor seems to hold it's state for ~30 seconds; and I receive advertisements in quick succession in response to any state change.  The motion (`data[7] & 0b10000000`) and contact (`data[7] & 0b00010000`) flags are what I needed, but there is other data there too.

Maybe @donavanbecker can help illuminate the format and describe why only the `SCAN_RSP` service data is used in [`node-switchbot`](https://github.com/OpenWonderLabs/node-switchbot/blob/d8a25e86a5fcdfafaa254950286fb438b08443dd/lib/switchbot-advertising.js#L94.

donavanbecker, 2022-12-16:
@dcousens I am not sure. I actually don't work for SwitchBot. I just maintain the nodejs repos. If someone wants to enhance node-SwitchBot, I would be happy 😀 to merge the pull request!

dcousens, 2022-12-16:
Interesting, where did the documentation come from in this repository?
Maybe @Minsheng could help?

bdraco, 2022-12-16:
@dcousens thats good news. How do you determine it's a contact sensor with only passive scans?

bdraco, 2022-12-16:
Also do you have any docs / code to parse the manufacturer data ?

dcousens, 2022-12-16:
@bdraco I determine the sensor from it's first appearance on powering on, with the company identifier `0x0969` for the [manufacturer data](https://github.com/OpenWonderLabs/python-host/blob/0b306eca80c468677672a90e25f38e7d154963c8/switchbot.py#L46).

I don't have other Switchbot products for now, so I don't know if you can reasonably disambiguate them if you're thinking of automation.

> Also do you have any docs / code to parse the manufacturer data ?

Maybe I'll create a pull request for @donavanbecker as suggested

bdraco, 2022-12-16:
> but I don't have any documentation other than what I have determined myself.

Looks like we are all in the same boat on this.  

Having docs for the manufacturer data would be a major improvement for the 3000+ users we have using the Home Assistant integration.

donavanbecker, 2022-12-16:
> Interesting!  Out of curiosity, where did you source the documentation in this repository from?

This all came from the wiki that was on the python-host repo. Just moved for easier access and updating.

bdraco, 2022-12-16:
https://github.com/Danielhiversen/pySwitchbot/pull/152

I can confirm the `manufacturer_data` gets sent passively!

I figured out 4 of the values in live testing.   

```python3
        motion_detected = bool(mfr_data[7] & 0b10000000)
        contact_open = bool(mfr_data[7] & 0b00010000)
        button_count = mfr_data[12] & 0b00001111
        is_light = bool(mfr_data[7] & 0b01000000)
```

I still don't have a way to identify the specific device though.

bdraco, 2022-12-18:
I managed to write `manufacturer_data` parsers for most of the devices pySwitchBot supports

It looks like the battery data isn't there though.

They are all there if anyone wants to lift them : https://github.com/Danielhiversen/pySwitchbot/tree/master/switchbot/adv_parsers

donavanbecker, 2022-12-20:
@bdraco Wanna create a PR for node-SwitchBot? :)

bdraco, 2022-12-20:
> @bdraco Wanna create a PR for node-SwitchBot? 😀 

If I didn't have a giant backlog of other work to get done I'd be happy to do that.  Sadly I'll be digging myself out of the hole for a while since adding the Bluetooth integration to Home Assistant has taken up about 10x more work than I expected 🙈

donavanbecker, 2022-12-21:
> If I didn't have a giant backlog of other work to get done I'd be happy to do that.  Sadly I'll be digging myself out of the hole for a while since adding the Bluetooth integration to Home Assistant has taken up about 10x more work than I expected 🙈

Totally understandable, could you update this documentation? For people to better follow?

dcousens, 2022-12-31:
@bdraco from my recent testing, `contact_open` isn't a bit flag.

```
contact_open = (mfr_data[7] & 0b00110000) >> 4
```

Where `1` is open, and `2` is open for greater than 60 seconds.

bdraco, 2023-01-12:
It looks like now have parsers for all the manufacturer data values in pySwitchbot so I'm going to close this.

Thanks to everyone who provided input.

closed 2023-01-12 by bdraco