OpenWonderLabs/SwitchBotAPI-BLE #16: SwitchBot Blind Tilt

donavanbecker, 2023-01-07:


bdraco, 2023-01-12:
There is a PR in progress https://github.com/Danielhiversen/pySwitchbot/pull/181 that might be helpful to provide some insight

AdyRock, 2023-02-24:
Is there a reason why the position information is now only available by BLE request instead of being sent in the Service Data section of the Advertisement, like it is for the curtains?
Having it advertised in the service data is a perfect system as it means no lengthy BLE requests need to be made and an near instant response is just there.
I can't really see it is a power saving feature as the advertainments are being sent anyway with the battery data.

Would it be possible to make it the same a the Curtain device as that works so well?

ndbeals, 2023-04-20:
Could we get some documentation on packet of the basic information response? pySwitchbot is currently reporting incorrect tilt data from a `get_basic_info()` (0x5702) request. Interesting to note, listening to advertisements and parsing the advertisement data from the blind tilt does give the correct tilt value.

AdyRock, 2023-04-21:
So, are you saying that you are seeing the blind tilt position in the advertisement data?
If so where are you finding that?

AdyRock, 2023-04-21:
OK, so update my Blind Tilt to version 2 firmware and I now see the position data in the Manufacturer data. So, no longer need a time consuming BLE command to get the position. Thanks to the SwitchBot team for doing that.

Could the format of the Manufacturer data be published to make it easier to decode all the information?

ndbeals, 2023-04-24:
@AdyRock here's code documentation from pySwitchbot of the advertisement data: https://github.com/Danielhiversen/pySwitchbot/blob/34125596d12a94a052cbcb6701778c48a20f8a83/switchbot/adv_parsers/blind_tilt.py#L5

donavanbecker, 2023-04-25:
If someone builds the doc, I will accept it as a PR on here.

donavanbecker, 2023-04-25:
@ndbeals I don't work for SwitchBot. I just maintain these repos for them.

donavanbecker, 2023-04-25:
@ndbeals This is the official SwitchBot organization, but they only monitor when the have updates from what I have experienced.