OpenWonderLabs/SwitchBotAPI-BLE #25: [HELP] SwitchBot Bot BLE API calls working anymore.

peterneutron, 2023-05-19:
### Current Situation

I ordered my first SwitchBot Bot in 2021 and use it to power on/off my coffee maker. Last week I ordered a second Bot to use with another device. Both Bots are on firmware V6.3.

I never use the official iOS app except to update the firmware but a custom `Python` script (see Configuration) instead to initiate a button press. The old Bot works absolutely fine with this and presses the button without issues but **NOT** the new bot. The weird thing is the script successfully writes the command to the UUID of the new Bot but does absolutely nothing until I open the app and tap into the settings view of the Bot, then somehow the Bot executes the apparently cached command send via the Python script. I can't make any sense of this so maybe someone here can provide more informations about this behaviour.

**UPDATE 1**

After playing with the BLE command responses there are differences between the Bot bought in 2021 and the new Bot (2023) . See Logs.

### Logs

```markdown
Bot 2021:

Connected: True
Service: 0000fee7-0000-1000-8000-00805f9b34fb
  Characteristic: 0000fec8-0000-1000-8000-00805f9b34fb
    Handle: 10
    Properties: ['indicate']
    Value: bytearray(b'indicate char')
  Characteristic: 0000fec9-0000-1000-8000-00805f9b34fb
    Handle: 15
    Properties: ['read']
    Value: bytearray(b'\xd4\xa6\xf6\xa4\xee!')
  Characteristic: 0000fec7-0000-1000-8000-00805f9b34fb
    Handle: 13
    Properties: ['write']
    Value: bytearray(b'write char')
Service: 00001801-0000-1000-8000-00805f9b34fb
Service: cba20d00-224d-11e6-9fb8-0002a5d5c51b
  Characteristic: cba20002-224d-11e6-9fb8-0002a5d5c51b
    Handle: 21
    Properties: ['write-without-response', 'write']
    Value: bytearray(b'W\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
  Characteristic: cba20003-224d-11e6-9fb8-0002a5d5c51b
    Handle: 18
    Properties: ['notify']
    Value: bytearray(b'\x01_@d\x00\x00\x00\xbd\x00\x00\x00\x00\x00')

---------------------------------------------------------------------

Bot 2023:

Connected: True
Service: 00001801-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a05-0000-1000-8000-00805f9b34fb
    Handle: 2
    Properties: ['indicate']
    Value: Read not permitted for this characteristic
Service: cba20d00-224d-11e6-9fb8-0002a5d5c51b
  Characteristic: cba20002-224d-11e6-9fb8-0002a5d5c51b
    Handle: 15
    Properties: ['write-without-response', 'write']
    Value: Read not permitted for this characteristic
  Characteristic: cba20003-224d-11e6-9fb8-0002a5d5c51b
    Handle: 17
    Properties: ['notify']
    Value: Read not permitted for this characteristic
```

### Configuration

```json
import asyncio
from bleak import BleakClient

# Replace with your SwitchBot device's MAC address
switchbot_mac_address = "XX:XX:XX:XX:XX:XX"

async def press_button(address):
    async with BleakClient(address) as client:
        # UUID for the RX characteristic of the message from terminal to device
        rx_characteristic_uuid = "cba20002-224d-11e6-9fb8-0002a5d5c51b"

        # Command bytes to send (0x57, 0x01, 0x00)
        cmd_execute_action = bytearray([0x57, 0x01, 0x00])

        await client.write_gatt_char(rx_characteristic_uuid, cmd_execute_action)
        print("Button pressed.")

loop = asyncio.get_event_loop()
loop.run_until_complete(press_button(switchbot_mac_address))
```

### Environment

- OS: Arch Linux ARM
- Python: 3.11.3

closed 2023-05-31 by github-actions[bot]

aki77, 2023-06-01:
This problem has not yet been resolved.

peterneutron, 2023-06-01:
Someone on Reddit helped me out with this, see [Link](https://www.reddit.com/r/TrySwitchBot/comments/13l2kof/comment/jlmgaop). It appears one has to subscribe to the Bots notification characteristic to get it to react to the BLE commands. This also seems to only affect newer Bots and iirc isn't documented anywhere in the BLE API documentation, if I'am wrong on this please point me to the correct docs.

Here is my revised python script to include the notification characteristic:

```
import asyncio
from bleak import BleakClient, BleakError

# Replace with your SwitchBot device's MAC address
switchbot_mac_address = "XX:XX:XX:XX:XX:XX"

async def press_button(address):
    try:
        async with BleakClient(address) as client:
            # UUID for the RX characteristic of the message from terminal to device
            rx_characteristic_uuid = "cba20002-224d-11e6-9fb8-0002a5d5c51b"

            # Notification characteristic UUID
            notify_characteristic_uuid = "cba20003-224d-11e6-9fb8-0002a5d5c51b"

            # Descriptor UUID
            descriptor_uuid = "2902"

            # Enable notifications
            await client.start_notify(notify_characteristic_uuid, notification_handler)

            # Command bytes to send (0x57, 0x01, 0x00)
            cmd_execute_action = bytearray([0x57, 0x01, 0x00])

            await client.write_gatt_char(rx_characteristic_uuid, cmd_execute_action)
            print("Button pressed.")

            # Notify for a second and then stop
            await asyncio.sleep(1.0)
            await client.stop_notify(notify_characteristic_uuid)

    except BleakError as e:
        print(e)

def notification_handler(sender: int, data: bytearray):
    print(f"Received data ({data}) from {sender}")

loop = asyncio.get_event_loop()
loop.run_until_complete(press_button(switchbot_mac_address))

aki77, 2023-06-01:
@peterneutron 

Thank you for sharing your solution!