OpenWonderLabs/SwitchBotAPI-BLE #26: Outdoor Meter BTLE API

rspier, 2023-06-10:
### Current Situation

Is the documented API at https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/meter.md correct for the Outdoor Temperature/Humidity Sensor (Model W3400010)

The broadcast messages don't seem to fit the schema in the doc.

### Logs

```markdown
Bluetooth Advertisement Data:

Length: 0x02, Type: 0x01, Data: bytes('06')
Length: 0x0F, Type: 0xFF, Data: bytes('6909XXXXXXXXXXXX360302963700')
Length: 0x06, Type: 0x16, Data: bytes('3DFD770064')
```
```

### Configuration

```json
N/A
```

### Environment

N/A

vostok92, 2023-06-10:
Based on my observations, it seems to be formatted as follows:
```
Length: 0x0F, Type: 0xFF, Data: bytes('6909XXXXXXXXXXXX360302963700')
  => 11-13th bytes("029637") indicates temperature and humidity:
         temp = ((data[10] & 0b00001111) * 0.1 + (data[11] & 0b01111111)) * (((data[11]&0b10000000)>0:1:-1);
         humi = data[12] & 0b01111111;
    i.e. 029637 -> temp = 22.2 (deg), humi = 55 (%)
 
Length: 0x06, Type: 0x16, Data: bytes('3DFD770064')
  => 5th byte ("64") indicates remaining battery:
    i.e 64 -> 100 (%)
```

rspier, 2023-06-11:
@vostok92 Thank you!

donavanbecker, 2023-06-11:
Please update documentation if you can.

rspier, 2023-06-11:
@donavanbecker I'd be happy to, but it would be nice to get the official schemas so there's not a lot of "Unknown".  This device seems to be significantly different in how it sends things from the ones documented at https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/meter.md

DigiH, 2023-06-11:
I found that the calculations were actually the same as the other SwitchBot Meters, apart from temp and hum shifiting to manufacturerdata and shifting the positions there, and battery remaining in servicedata with occupying the last octet.

From user feedback I found that these seem to be correct.

But I totally agree, some officially sourced documentation would be the best solution for all.

donavanbecker, 2023-06-12:
@rspier I don’t actually work for SwitchBot so can’t get that for you. I just manage the repos for them.

donavanbecker, 2023-06-12:
That is why I was hoping the community would put it together until SwitchBot releases something.

closed 2023-06-13 by donavanbecker