OpenWonderLabs/SwitchBotAPI-BLE #29: Add information for Thermometer Hygrometer Indoor Outdoor

AdyRock, 2023-06-30:
### Current Situation

There are no details documented for the BLE data format of the Thermometer Hygrometer Indoor Outdoor

### Proposed Change

Add the BLE data format information for the Thermometer Hygrometer Indoor Outdoor

### Additional Context

I would like to add support for this device

AdyRock, 2023-06-30:
I have cracked it.
The device Type is 'w' and the temperature and humidity data is in the Manufacture data block as follows:
```
        const tempSign = (byte11 & 0b10000000) ? 1 : -1;
        const tempC = tempSign * ((byte11 & 0b01111111) + ((byte10 & 0b01111111) / 10));
        const humidity = (byte12 & 0b01111111);

```
Battery is still in the service data as for other devices.

donavanbecker, 2023-06-30:
https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/issues/26