OpenWonderLabs/SwitchBotAPI-BLE #33: Document Manufacture Data for "T" device type (i.e. WoSensorTH)

moritzmhmk, 2023-09-06:
### Current Situation

Currently the documentation focusses mainly on the Service Data of the "T" device type.

### Proposed Change

The following shows the Manufacturer Data and Service Data of the "w" and "T" device type (i.e. WoIOSensorTH and WoSensorTH).

```
  <       Co.ID |    MAC Address    | CC ?? | SensData |    > <       DevData  | SensData >
w <Buffer 69 09 | xx xx xx xx xx xx | 2c 0b | 00 95 4a | 00 > <Buffer 77 00 64 >
T <Buffer 69 09 | xx xx xx xx xx xx | 29 03 | 05 98 38 >      <Buffer 54 00 e4 | 05 98 38 >
```

The Manufacturer Data is as follows for both "w" (Outdoor Meter) and "T" (Indoor Meter) device types:
- Co.ID: Company identifier 0x0969
- MAC Address
- CC: Change Counter - increases when a value changed from previous broadcast
- ??: Byte of unknown meaning
- SensData: Sensor Data (i.e. temperature and humidity) as currently documented
- type w only: trailing 0x00

The Service Data is identical for the first 3 bytes (as already documented) and for the type "T" includes the same SensData as in the ManufacturerData.

### Additional Context

Maybe someone can confirm if this is also true for the "i" device type (Meter Plus).

Is there any documentation on the byte marked "??" above?

donavanbecker, 2023-09-17:
Please open PR if this is correct so it can be added

moritzmhmk, 2023-09-17:
I am not entirely sure if this is dependent on the firmware version or hardware revision.  Some feedback from other users before adding this would be great.

orcagami, 2023-12-10:
I can confirm the manufacturer and service data of Meter Plus ("i") are identical to what is reported here for the indoor meter ("T").
The unknown byte seems to be consistent from my limited testing. My outdoor sensor ("w") also reports it as 0x0b while for both my Meter Plus devices it is 0x0f.

DigiH, 2024-01-01:
I've seen Outdoor Meters, even the same device, reporting **0x03** or **0x0b**, and **0x0f** for a different Outdoor Meter, so possibly it is that bit[0] and bit[1] are always true, with bit[2] and/or bit[3] being true **or** false.

donavanbecker, 2024-01-01:
Please open a PR if you want to update the documentation

DigiH, 2024-01-01:
As this is only the observation I have made from several different Outdoor Meters, and as **0x0f** or **0x03** at least seem to be possible with the Outdoor Meter **as well as** Meter Plus and other devices, I'm not quite sure what to submit ;)

closed 2026-10-30 by donavanbecker