OpenWonderLabs/SwitchBotAPI-BLE #40: Lock Pro BLE API

schummar, 2024-06-03:
### Current Situation

There is no local integration (Home Assistant etc.) at the moment to work with the new Lock Pro.

### Proposed Change

I would be very helpful to see the BLE API docs for the Lock Pro

schummar, 2024-07-22:
Any chance to get some docs soon? The home assistant integration is kind of working, but there are a lot of educated guesses involved - which means some things simply don't work right yet.

schummar, 2024-08-20:
Allow me one more ping. One particular thing that currently doesn't work, is the detection of whether the lock has the night latch feature. It would already be immensely helpful to know what bit contains this information. So even if a full documentation is not there yet, could you provide this piece of information?

steefan85, 2024-11-20:
Also looking forward to a Lock Pro BLE API. I'd like to integrate the Lock to iobroker; there is still no adapter available because the API doesn't exist.

closed 2026-10-30 by donavanbecker